OrcaSlicer/OrcaSlicer #576: [Feature Request] Add custom printer via wizard

esc247, 2023-03-24:
**Is your feature request related to a problem? Please describe.**
Currently if you want to add a custom printer you have to create the json files and then set it up from there. It would be great to be able to set up a new custom printer using the GUI.

**Describe the solution you'd like**
From the printer selection section, if there was a button to "Add Printer" from which you could set up a printer, that would be great.

ramaxa9, 2023-04-02:
i used one of profiles **OrcaSlicer_V1.5.0_Win64\resources\profiles** as example, i made a copy to **AppData\Roaming\OrcaSlicer\system** as well
but wen change inheritance of **AppData\Roaming\OrcaSlicer\user\default\machine\FlyingBear Ghost 5.json** to profile i made my,  my profile not showing in the profile list

esc247, 2023-04-09:
Hi, you have to have the correct paths and filenames defined in the printer_name.json file.

martinrosolino, 2023-05-28:
hey! im stuck, ive created all the json files ...\resources\profiles, then copied them to AppData\Roaming\OrcaSlicer\system following the example of the existing ones, but the printer its not there when i want to add it once inside the slicer

closed 2023-11-20 by github-actions[bot]