OrcaSlicer/OrcaSlicer #577: Max volumetric speed test doesn't go above 20mm3/s

auradata, 2023-03-24:
If you set up a calibration of max flowrate with a max rate over 20mm3/s, it creates a model the correct height but any height which should be above 20mm3/s is only showing flow rates of 20.58 in the slicer

Prutsium, 2023-03-24:
Check the max flow setting in the Filament at that is default 20.
[image]

auradata, 2023-03-25:
The calibration script automatically sets that to 200, I've tried setting it manually too but it still won't go above 20 in the test print

Prutsium, 2023-03-25:
Strange i can push it over 60
[image]

auradata, 2023-03-25:
The part was set to the wrong filament which still had a 20mm3/s limit

closed 2023-03-25 by auradata