OrcaSlicer/OrcaSlicer #578: Autoarrange object placement

rkarlsba, 2023-03-24:
If I load three objects and press A, the objects are placed in the upper right corner
[image]

Pressing shift+A, however, places them in the centre, which I've always seen before
[image]

Any reason of this change? I can't see why placing them in a corner will do anything good.

This is on OrcaSlicer 1.5 on mac intel

bojangles86, 2023-03-26:
odd. i can load 5 objects on the plate and arrange them and they all sort themselves in the middle (more or less). 
orca 1.5 on linux.

rkarlsba, 2023-03-26:
I dug a bit further into this and found this: I started a new project and added two elementaries, cubes. Pressed a and they autoarrange in the middle. Then, if I change one part to using tree supports (which it obviously doesn't need, but the other parts earlier, did), both cubes gets placed in the corner with a.

closed 2023-11-20 by github-actions[bot]