OrcaSlicer/OrcaSlicer #579: unable to select detect thin walls

unwritten-ferret, 2023-03-24:
**Describe the bug**
default 0.2 profile with 0.4mm nozzle the `detect thin walls` checkbox in strength is not active and cannot be used. tried multiple models and presets. in bambustudio this box is always active and checkable

**3mf File for This Bug**
n/a
all models i try

**To Reproduce**
load model, try to select `detect thin walls` under strength

**Expected behavior**
checkbox active and usable

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Printer model**
bambu x1c

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows] windows 10 x64
 - Version [e.g. 22] orca 1.5.0

hliebscher, 2023-03-25:
it's not a bug it works only in classic mode not arachne!

crushedpadding, 2023-04-09:
@hliebscher probably close since not a bug

closed 2023-11-20 by github-actions[bot]