OrcaSlicer/OrcaSlicer #580: Export Config stop working on V1.5.0

TechDude23, 2023-03-24:
With Softfever 1.4 you could export your configs but after installing Orca silicer 1.5 you can not export any new configs made. 

If you have any past configs with same name in the folder then it will overwrite and update that profile fine.

My work around for now has been to use soft fever until 1.5 is fixed for exporting my configs. 

It would be nice to also have a option to ( export all configs ) for backing up all your profiles with just one click.  

Thank you for the great slicer :)

TechDude23, 2023-03-24:
Window 64 version

superceleron, 2023-03-29:
i can confirm this it does not export or import, on 1.5.0.

markuswarlich, 2023-03-30:
i can also confirm. It does not export or import, on 1.5.0.

DanielGroot, 2023-04-01:
i can also confirm. It does not export or import, on 1.5.0.

TechDude23, 2023-04-20:
Thank you for Fixing it

closed 2023-04-20 by TechDude23