OrcaSlicer/OrcaSlicer #582: "Only 1 wall on top surfaces" only works for top-most surface

npoulos, 2023-03-25:
Tested in regular bambu slicer vs orca and it only seems to be an issue in orca.

If you add a primitive cube it will slice correctly with 1 wall at the top. but if you add a any model with more than 1 top surface, onnly the very top surface in the model has 1 wall while the other top surfaces keep whatever the wall loops is set to.

You can test this with make 2 cubes, one bigger than the other, assemble them and then slice.

npoulos, 2023-03-25:
[image]

hliebscher, 2023-03-25:
is open here again #547

liftbag, 2023-03-26:
> here

And here #379

npoulos, 2023-04-04:
@SoftFever any thoughts on fixing this one? The issue doesn't seem to appear in BambuStudio

dantetemp, 2023-04-05:
@SoftFever any updates on this?

npoulos, 2023-04-29:
Hi @SoftFever is there a reason all the tickets re: this issue are being ignored?

SoftFever, 2023-04-29:
Hold your horses, boys.
I'm not a super man

SoftFever, 2023-04-29:
FYI, this is an issue for the Arachne generator only.
You can switch to "Classical wall generator" if you want a true 1 wall top surface for now.

npoulos, 2023-04-29:
Woops! Sorry to put pressure on you! Just wasn’t sure if these were being “ignored” or just not yet gotten to! Thanks for all you do mate 😊 will sit tight!

Noisyfox, 2023-07-22:
> @SoftFever any thoughts on fixing this one? The issue doesn't seem to appear in BambuStudio

If you look more closely you'll notice that when Arachne is selected the "Only one wall on top surfaces" options is disabled in BBS.

closed 2023-11-20 by github-actions[bot]