OrcaSlicer/OrcaSlicer #590: [Bug] OrcaSlicer Crashes when trying to embed text

esc247, 2023-03-26:
**Describe the bug**
When you use the add text tool in OrcaSlicer, you cannot embed the text. As per the screenshot: 
[image]

**To Reproduce**
After adding an object, select the Text embed tool and set the embedded depth value. Changing the value has no effect.

**Expected behavior**
When setting the value of the embedded depth, I was expecting that the text would be embedded into the object, however it does nothing.  I have added screenshots of thickness value changed as well. The min value that is automatically set is 0.10, even if I put in a negative value.

**Screenshots**
[image]
[image]

**Printer model**
VzBoT330

**Desktop (please complete the following information):**
 - OS: Mac
 - Version 13.2.1

Cactus1906, 2023-03-30:
I don't know how this should work. 
Maybe the following is a workaround.
Right click on the "text_shape" in "objects view" and "Change Type" to "Negative Part" then the text is embossed.
In addition you can move the object "into" the object. In my example it's a 50mm cube with 9 mm thick negative text.
Hope it helps.
[image]
[image]

esc247, 2023-04-02:
Thanks will try that out

esc247, 2023-04-05:
I tried this, but my I still does not do what you did.
[image]
[image]
[image]

esc247, 2023-04-05:
Oh wait.... It worked! I had to slice it to see it :) Thanks very much!

closed 2023-04-05 by esc247