OrcaSlicer/OrcaSlicer #591: Custom Variables

wiggywoosaa, 2023-03-26:
It would be nice to have custom variables like superslicer has in the 'filament section - notes - custom variables'

TexZeTech, 2023-04-06:
[image]

something for macros to use. 

@wiggywoosaa You should add your use case and add a screen shot at the least.

voidtrance, 2023-08-23:
I second this request.

In my case, I assign a unique ID to every filament brand that I can then use to uniquely identify the filament in my macros and set various settings accordingly.

kylek29, 2023-08-25:
I think it'd be a very useful feature port. Can we also get the Filament Notes section? Seems like it'd be useful for keeping those oddly specific filament tips/notes in a good place.

closed 2023-12-01 by github-actions[bot]