OrcaSlicer/OrcaSlicer #592: [bug] P1P Aux Fan not responding to Orcaslicer files.

bonjipoo, 2023-03-26:
If I set aux fan speed in filament setting on Bambu slicer the aux fan runs at the set speed during printing as expected.

If I take that file and open it in orcaslicer and print the file with aux fan speed set in the same way, the aux fan doesn't run at all during the print as if the fan is disabled in orcaslicer.

Not sure if this is a bug or if i'm missing something?  The aux fan does work on my P1P and if you set a fan speed via handy app it works as expected.

Kaden116484, 2023-03-27:
Hello, I am also having the same issue, aux fan works but not truing on during printing.

SoftFever, 2023-03-30:
You will need to enable `Auxiliary part cooling fan` in Printer settings first.

closed 2023-03-30 by SoftFever

Brandon520x, 2023-08-14:
@SoftFever I dont think this should be closed as I am also having the same issue wit the current (as of a few days ago) release. 

I have a PLA filament defined with Aux fan speed set to 100%
I have my QIDI X-Plus 3 printer defined with Aux fan checked

Inspection of the GCode shows the M106 P2 255 command is not being issued.  When I compile GCode from  the QIDI slicer (based on prusa slicer) it will turn on the aux fan, and the exhast fan (p2 and P3).  

I have noticed problems with the QIDI chamber heater not working if you define heating parameters in your start code, so I tried to disable these and the Aux fan will still not work.

Brandon520x, 2023-08-14:
@SoftFever I found a workaround.  The PLA filament profile i created was cloned from a profile that was for my old XMax which did not have a part cooling fan.  When i recreated the profile from one of the generic PLA profiles that were for my X-Plus 3, the fan started working.  This is weird since ALL of the setting are the exact same.  There mush have been something hiding behind the user settings that was preventing the fan from turning on.