OrcaSlicer/OrcaSlicer #594: Identify Completely Floating Regions

diginfo, 2023-03-27:
It would be great if as well as this warning, it actually identifies the problem by maybe indicating it with a colour and zooming into the layer, or something similar.

I am getting this error and I cannot find where the problem is, might be that I just don't know how to see it.
[image]

[image]

Ltek, 2023-10-03:
Agreed, I got this error today but cant find anywhere that it is floating... I'm suspecting this is a bug but if it is identifying a real issue with the design, it should colorize it so we know where.

pacmac, 2024-01-02:
It's not stale, the issue still exists but is probably not a priority.

stephbellemare, 2024-01-30:
I found that if you slice and it gives a floating warning, change the infill %, reslice and put back to original infill %, slice again, the warning will have disappeared.

Ltek, 2024-01-30:
I find the warning has been consistent (and correct) for me... but would like the slicer to color-code the onscreen image so we can see what it has identified. Its often very difficult to see what it thinks is floating vs just a normal overhang

ThrawnyBoy, 2024-02-02:
> I found that if you slice and it gives a floating warning, change the infill %, reslice and put back to original infill %, slice again, the warning will have disappeared.

This worked for me so definitely a bug somewhere. Thanks for the tip to workaround the error it was annoying me when you can plainly see (and _know_) there is absolutely no floating regions.

MaDMaCKeM, 2024-02-28:
Same here, changed infill setting and the error disappeared

Ltek, 2024-05-30:
ping

Ltek, 2024-09-01:
not stale

TomasL-SE, 2024-10-20:
Got this error as well, added support to se what happens, and the support ended up one layer above the top layer. 
This does not make sence

Ltek, 2025-01-19:
still needed!

GreavesDesignerOne, 2025-03-22:
Encountered "floating regions" error in [this IDEX calibration model](https://www.printables.com/model/201707-x-y-and-z-calibration-tool-for-idex-dual-extruder), it's a very simple model and there's no issue slicing it in Cura.

closed 2025-06-29 by github-actions[bot]

boksbox, 2025-09-16:
Still persists at 2.4.1 beta.   

Shows warning when no visible floating regions and slice looks fine.  Prints ok.

[image]

Warning disappears when supports enabled but no supports generated

[image]