OrcaSlicer/OrcaSlicer #599: Orca slicer corrupts 3mf file saved by Bambu slicer without warning

wbphelps, 2023-03-28:
**Describe the bug**
Opening a file saved by Orca Slicer (OS) with Bambu Slicer (BS) will generate a warning message. the reverse is not true.  If a BS file is opened by OS and saved, the file can no longer be used by BS.  There is no way that I know of to reverse this other than resorting to a backup file

**3mf File for This Bug**
I can append a 3mf file that was damaged but it's very easy to create one.

**To Reproduce**
Open a file saved by BS, and save it.  There are no warning messages.  File opened by BS will load geometry only, any custom settings are lost.

**Expected behavior**
Ideally OS would use a new filetype, or at least have that option.  Given that this also happened going from Prusa to Bambu, I don't expect this to happen.  One can hope.  An attepmt to save a file with Orca that was created by Bambu should cause a warning message BEFORE the save is done.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Printer model**

**Desktop (please complete the following information):**
 - OS: WIndows 11
 - Version current

hliebscher, 2023-03-28:
this is not a bug ! Orcaslicer has many more settings at the 3mf for all features thas BS not have.
use save as ... than have you 2 versions

wbphelps, 2023-03-28:
I think it is a bug that Orcaslicer will load a Bambu slicer file and save it with no warning message about compatibility. That's all I'm asking for is some warning message so I don't lose days of work again.

SoftFever, 2023-04-01:
@wbphelps Point taken.
I will add a new feature to save to the BambuStudio-compatible project format in the future.

SoftFever, 2023-04-05:
fixed in 1.60

closed 2023-04-05 by SoftFever