OrcaSlicer/OrcaSlicer #603: [Feature Request] Add X-Y hole compensation on object/parts/modifier level

Launert, 2023-03-28:
**Is your feature request related to a problem? Please describe.**
For multi material printing, different X-Y hole compensation values are needed for perfect accuracy for all parts.

**Describe the solution you'd like**
Add X-Y hole compensation to the object view within Orca Slicer to allow different parts of the model to have different settings.
[image]

**Describe alternatives you've considered**
Shrinkage compensation on filament settings is a good step, but doesn't help if the outer dimensions of the object are already good.

**Additional context**
Add any other context or screenshots about the feature request here.

aleung, 2023-05-07:
I see this option in 1.6.2 Linux version.

But the X-Y hole compensation seems to be wrong. The description says "Positive values make holes bigger" but actually positive value makes holes smaller.

[image]

knw257, 2023-09-08:
I would also like this feature, to be able to edit the X-Y hole compensation on only part of a model, rather than the whole thing.

@aleung While you can change the X-Y hole/contour compensation at the part level, adding a modifier object to a part does not give this option. I believe this is what OP was requesting, and what I would like to see as well

TexZeTech, 2023-12-20:
Just went to use the modifier object to tweak the size of one hole and found this to not be there, looking forward to when this feature is added.

closed 2024-03-27 by github-actions[bot]