OrcaSlicer/OrcaSlicer #607: Viewing the mesh in Paint mode

Kyrian-29, 2023-03-29:
When using "Paint" with the "Triangle" tool, have the possibility to visualise the mesh of the part to be coloured.

SoftFever, 2023-03-30:
Can you elaborate more? 
Screenshots or recordings are helpful

Kyrian-29, 2023-03-30:
Trimming with the triangle tool is not easy without being able to clearly see the mesh (the triangles).
[image]

A view of the mesh like with NetFab for example would be easier to do a perfect trimming.
[image]

cybrsage, 2023-07-05:
I third the request, and second the thanks!!!

Noisyfox, 2023-09-02:
BBS used to have this feature but then they decided to disable this due to some macOS performance issue:

https://github.com/bambulab/BambuStudio/commit/8b5a49f37870b25623d75336d46065653bd645cc

PR #2808 (2023-11-21): Show wireframe in MMU painter gizmo
Now you can press Alt+Shift+Enter to show wireframe in MMU paint gizmo:

[image]

Adopted from BBS

Fix #607
closed 2023-11-21 by SoftFever