OrcaSlicer/OrcaSlicer #608: Add the option to select which filament is used for flow calibration in multi-filament prints

kvnper, 2023-03-30:
[Note: I raised this same issue in the official slicer, but will repost here just in case - ignore if not acceptable]

Please add the option to select which filament is used to calibrate filament flow.

I have a project where I am using a different material for the supports, and I'm also using a raft with that material because the contact area of the part to the bed is minuscule. The issue I am having is that the support filament is being used as the flow calibration filament.

Please see the photo below, where the yellow filament is the support material and is being used for filament flow calibration:

[image]

kvnper, 2023-03-30:
No luck with that unfortunately. My workaround at the moment that works for me is just to print a flat cube with the main filament in order to get auto calibration.

caption: filament 2 is still used, settings included in screenshot
[image]

SoftFever, 2023-03-30:
It looks like no black material is used for the first layer, which may explain it.
But yeah, a small sacrifice block should do the trick.(You can make it only 1 layer high to save filament)

closed 2023-11-20 by github-actions[bot]