OrcaSlicer/OrcaSlicer #611: Implement hotfix for https://github.com/bambulab/BambuStudio/releases/tag/v01.05.00.61

Emperornero, 2023-03-30:
**Describe the bug**
A clear and concise description of what the bug is.

3rd party logins can have an issue with sending prints to printers over the cloud or network.

**To Reproduce**
Steps to reproduce the behavior

Unsure.  Experienced this just now.  Had to disconnect printer from my account, reconnect it, and then log in and out of Orca Slicer to print over Cloud/Network again.

**Expected behavior**
A clear and concise description of what you expected to happen.

The print is sent and then the page is moved to the Device tab.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Printer model**

X1-Carbon

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]

SoftFever, 2023-03-30:
OrcaSlicer 1.5 includes this hotfix

closed 2023-03-30 by SoftFever

Emperornero, 2023-03-30:
> OrcaSlicer 1.5 includes this hotfix

I am on 1.5 and just experienced this. 

Should I be wiping the entire folder when upgrading, or is only overwrite good enough?