OrcaSlicer/OrcaSlicer #612: Snapmaker J1 Printer Profile

Learoy1, 2023-03-30:
Please can you add the Snapmaker J1 IDEX printer to the 3rd party list. 

Snapmaker Luban is just horrible, Cura is terrible at multicolour and crashes constantly and I'd love to be able to use Orca for all my printers.. 

Hopefully the info below is what is required.

Snapmaker J1 config files for PrusaSlicer
https://github.com/leandrolima-nyc/SnapmakerJ1

https://forum.snapmaker.com/t/great-sharing-prusaslicer-profiles-by-dylan-and-mcgybeer/28796

hliebscher, 2023-03-31:
i have this  running including postprocessor and send to printer

hliebscher, 2023-04-06:
wait for the PR idex J1  now working #663

Learoy1, 2023-04-06:
> wait for the PR idex J1 now working #663

Thank you I appreciate this so so much.

Learoy1, 2023-05-02:
Ok just tried to start a print sliced from orca and nothing happens.

closed 2023-11-20 by github-actions[bot]