OrcaSlicer/OrcaSlicer #615: 'Arrange all objects' does not take into account the possibility of placing the objects inside each other.

dojo90, 2023-03-31:
**Describe the bug**
I have the following model file and would like to place it on the print bed and print it as often as possible.

If I now duplicate the object and execute "arrange all objects", the slicer does not place them inside each other, but from the 4th part onwards a new plate is used.

[image]

[image]

**3mf File for This Bug**

[ArrangeBug.zip](https://github.com/SoftFever/OrcaSlicer/files/11120619/ArrangeBug.zip)

**Expected behavior**
If possible, the parts should be positioned as follows:

[image]

[image]

Ibot-11, 2023-07-16:
Prusaslicer does a much better job arranging objects. They improved the algorithm or something. Also it doesn't rotate the models by 45° anymore which causes that infill is not 45° anymore.

Maybe it’s an option to port the new arrangement algorithm from Prusaslicer?

Orca:
[image]

Prusaslicer:
[image]

Both with 1mm model distance and rotation enabled.

closed 2023-11-20 by github-actions[bot]

djonasdev, 2024-03-14:
Still relevant. Can you please reopen it?