OrcaSlicer/OrcaSlicer #616: Letters

renemosner, 2023-03-31:
**Describe the bug**
Please correct or add some letters, see  photo.  I am translating cs_CZ into Czech and these characters are unwanted.  Another question is whether it is possible to add pom_merger in the future as used by SuperSlicer to make translation easier.  Thank you and best regards, Mošner
( ě š č ř ž ů)

[image]
[image]

**Printer model**
Own construction

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version: OrcaSlicer v1.5.0_Win64

hliebscher, 2023-03-31:
you have to make a seperate directory cz with file Orcaslicer_cz.po based on Orcaslicer.pot
then translate all msgstr "????" into you language

renemosner, 2023-03-31:
Yes, I understand it.
But it is the font in the options on the left that is supported, but not in the places marked in the picture.
I'm working on the translation, but I can't visually check it.

hliebscher, 2023-03-31:
pls send your  cz.po file  the i check this with header you use

for  german ... the "Content-Type: text/plain; charset=UTF-8\n" is important !

msgid ""
msgstr ""
"Project-Id-Version: Orca Slicer\n"
"Report-Msgid-Bugs-To: \n"
"POT-Creation-Date: 2023-03-16 21:34+0800\n"
"PO-Revision-Date: \n"
"Last-Translator: Heiko Liebscher <hliebschergmail.com>\n"
"Language-Team: \n"
"Language: de\n"
"MIME-Version: 1.0\n"
"Content-Type: text/plain; charset=UTF-8\n"
"Content-Transfer-Encoding: 8bit\n"
"Plural-Forms: nplurals=2; plural=(n==1) ? 0 : 1;\n"
"X-Generator: Poedit 3.2.2\n"

renemosner, 2023-03-31:
Here it is.
all changed files are there. In the text.json file I changed the .de translation to cz just to test the menu translation.
Only the OrcaSlicer cs_CZ translation is not yet 100% complete.

[OrcaSlicer.zip](https://github.com/SoftFever/OrcaSlicer/files/11126881/OrcaSlicer.zip)

hliebscher, 2023-04-01:
we have to fix the truetype font settings ... and fix some other gui points 
i have no windows but i integrate this and than we add again you translation file

renemosner, 2023-04-01:
OK:
Thank you for your willingness and time.
I will look forward to the results.

hliebscher, 2023-04-05:
i have a pr #654

renemosner, 2023-04-05:
Yes, I know about that.
But it's only basic machine translation. I incorporated translations from PrusaSlicer and SuperSlicer into my translation for a better understanding of the user.
And as I wrote, it's not 100% finished yet.
And in the text.json file, the translation to the main menu is just under the (de .)heading. which can be used for cs localization.

renemosner, 2023-04-05:
Yes, I looked at it, but the characters are still wrong with letters with hooks and circles.  As I wrote in the original message.

renemosner, 2023-05-02:
Hello. 
 I want to ask how it will be with bad characters, I haven't noticed any improvement.  Will you solve it somehow in the future? 
 Thank you.

closed 2023-06-10 by renemosner