OrcaSlicer/OrcaSlicer #620: Changing the Extruder Clearance Radius does nothing

jeffamelang, 2023-04-01:
**Changing the Extruder Clearance Radius does nothing**

**To reproduce:**
1. Change the print sequence to "by object"
2. Put two objects _just_ close enough together so that the slicer says they're too close
3. Change the printer's "Extruder Clearance - Radius" to smaller, like 10mm
4. Notice that the slicer still says they're too close.
5. Change the "Extruder Clearance - Radius" arbitrarily and notice that nothing changes. 

**Possibly related**
Bambu Studio doesn't let you change the `extruder_clearance_radius`, but you can change it in `C:\Users\[yourName]\AppData\Roaming\` **BambuStudio** `\system\BBL\machine\fdm_bbl_3dp_001_common.json` and it'll use the changed value. When I try to change the value in `C:\Users\[yourName]\AppData\Roaming\` **OrcaSlicer** `\system\BBL\machine\fdm_bbl_3dp_001_common.json`, Orca overwrites that file each time it starts up so the setting doesn't stick.

**Expected behavior**
Reducing the clearance would make the slicer not say that the objects are too close. 

**Printer model**
Bambu X1C

**Desktop (please complete the following information):**
 - OS: Windows

[image]

[image]

SoftFever, 2023-04-02:
fixed.

jeffamelang, 2023-04-02:
woah, super fast. great job. it's refreshing when someone actually pays attention to reproducible bugs. i'll file others as i find them.

closed 2023-04-05 by SoftFever

rizwansarwar, 2023-04-24:
So at least I can't get it to do what I can do in Prusa Slicer. For example if I have objects stacked on top of each other, I can print them sequentially (or by objects as it is called in Orca slicer). However in Orca/Bambu, the extruder clearance keeps getting in the way. Even if I set the radius to 0 and height of rod to 200mm, it still complains about collisions. is it not possible to to do this in Orca slicer?