OrcaSlicer/OrcaSlicer #621: Voron: Post-processing scripts

H4kunaM4tata, 2023-04-02:
I am trying to make Orca Slicer the only slicer instead of having to use PrusaSlicer. I had X1C before and SoftFever was a blessing.
Atm, this is the main blocker.
There is a big difference between the times the slicer displays and how long the printer actually takes.
Some folks need this or any other post processing.

[image]

Script execution

Each script will be passed the absolute path of a temporary G-code file that PrusaSlicer generates. This file is stored in a temporary folder on your drive (typically your fast system drive). The script is then executed to modify the G-code in place and the resulting G-code file is written to your selected target folder or sent to a print host, such as PrusaConnect or Octoprint.

SoftFever, 2023-04-02:
Post-processing scripts have been supported since V1.4.3, and klipper_estimator should just work.
Please ref to #159

BTW, if you see a big-time estimation difference between the slicer and actual time, it is usually caused by inaccurate machine limit settings.(it's not as accurate as klipper_estimator but it is reasonably close)

H4kunaM4tata, 2023-04-02:
Linux version does not work the search feature button which kind sucks.
I was looking under the printer settings, found it under others.
Can run `klipper_estimator` again.

Thank you

closed 2023-04-02 by H4kunaM4tata

H4kunaM4tata, 2023-04-11:
@SoftFever I thought you should know, the search field not working on Linux I mentioned above, has nothing to do with OS but the main which I already knew, this is supposed to be fix with the coming releases.