OrcaSlicer/OrcaSlicer #622: Unable to Export Config Files

MiscPrinter, 2023-04-02:
**Describe the bug**
When trying to export the custom config file it silently fails. I have set the profiles to non-system profiles but the export function reports 0 profile exported as it only exports non-system files. It seems to treat all profiles as system files. I have tried with several printer, filament, and print profiles. A few days prior I was able to get some but not all of the configurations exported after restarting the app. Restarting the app no longer helps the config files export. 

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
File->Export->Export Configs ->prompt to select an output designation->open-> confirmation window shows that 0 config files were exported as only non-system files are exported. 

**Expected behavior**
File->Export->Export Configs ->prompts to select an output designation->open-> confirmation window shows that 3 config files were exported

**Screenshots**

[image]
[image]

**Printer model**

X1CC

**Desktop (please complete the following information):**
 - OS: MacOS
 - Version 1.5.0

SoftFever, 2023-04-02:
Fixed.
Will include the fix in next update

MiscPrinter, 2023-04-02:
Thanks for the great work.

SoftFever, 2023-05-21:
Fixed

closed 2023-05-21 by SoftFever