OrcaSlicer/OrcaSlicer #624: dark mode is messed up (probably after sleep)

domik82, 2023-04-02:
Hi,

This happened to me few times already so I guess it's worth reporting.
Dark mode seems to be messed up like on screenshots - I think it's happening after laptop is put in sleep. 

In my case I was planning to print 2 plates (each ~5-6h). So I scheduled first  - left laptop overnight with plan to schedule second print in the morning.

Slicer looked like on screenshots.
[image]

This is 1.5 version

Quick-Flash, 2023-04-03:
I get this same problem. Its an issue I have had on BambuSlicer as well. Its happened with every version of a BambuSlicer based slicer that I have used. So its not just an OrcaSlicer issue.

bdistin, 2023-04-20:
This also occurs when RDPing into the machine with the slicer.

aow99, 2023-05-25:
Same, noticed it on Bambu Studio a few months ago. Seems to happen after my pc is unlocked.

closed 2023-11-20 by github-actions[bot]