OrcaSlicer/OrcaSlicer #625: Shrinkage setting shrinks model even more!

umut885, 2023-04-02:
**Describe the bug**

The newly added "Shrinkage" option in filament setting shrinks the model. I tried 100% and 102%. I expected that the tolerance of 102% is higher to fit a round object into a cylindric hole.

100%: The round object fits (a little tight) into the cylindric hole
102%: The round object does NOT fit into the cylindric hole. The shrinkage option did shrink the hole instead adding more tolerance for an easy fit!

**Desktop (please complete the following information):**
 - OS: iOS intel Chip
 - Version 1.5.0 Based on BambuStudio 01.05.00.51

netweaver1970, 2023-04-02:
It's not a bug, it's as it's supposed to work. Read the tooltip for this parameter carefully. So a shrinkage value <100% will actually cause the part to be scaled UP to compensate. And scaled DOWN for shrinkage value>100%. Hence, the behaviour you see. It might be counter-intuitive, but that's the way the parameter is defined. It's also like this in SuperSlicer, in fact this option came straight from SuperSlicer, on special request.

[image]

The abover parameter is only for external perimeter/dimension compensation, for X-Y holes **inside** the part, one might need also the hole compensation
[image]

umut885, 2023-04-02:
Oh okay. Now I understand it. The tooltip takes really long time to appear. At least in my case. But now it makes perfectly sense. Also thank you for the hint regarding X-Y hole compensation. As mentioned before, without playing with X-Y hole compensation ( I hadn't seen this even until now) the Shrinkage option had definitely an impact on the hole size.

I will close the issue-

closed 2023-04-02 by umut885