OrcaSlicer/OrcaSlicer #627: ARC fitting is rare and inconsistent

BrettThePark, 2023-04-02:
**Describe the bug**
ARC fitting seems to be rarely used, in places where it should be fully used. Often in models I have circles which end up being drawn as line segments. I am not sure if this caused by a conflicting setting or if the fitting tolerance is too low (I am surprised the value cannot be adjusted within a profile, unless I missed it).

**3mf File for This Bug**
This is the 3mf file (change zip extension back to 3mf).
[SimpleCylinderArcFail.zip](https://github.com/SoftFever/OrcaSlicer/files/11133551/SimpleCylinderArcFail.zip)

**To Reproduce**
I created a cylinder in OrcaSlicer that is about three layers high. I set ARC fitting on with no top or bottom layers and a 15% infill (concentric).

**Expected behavior**
I expect all lines to be drawn using arcs.

**Screenshots**
Layer 1 uses G3 codes to draw the walls and infill
[image]

Layer 2 uses G1 codes for the walls, and G3 codes for the infill
[image]

**Printer model**
BambuLab X1C

**Desktop (please complete the following information):**
 - OS: MacOS 13.2.1
 - Version 01.05.00.61

npoulos, 2023-04-04:
I've found enabling Arachne uses less arcs if the line needs to be variable in width. Have you tested with Classic turned on?

BrettThePark, 2023-04-04:
After digging through the code it appears that the arc welding tolerance is set by the Resolution value in the UI (the tool tip talks about "simplifying" ... which from a users perspective does not make it clear that using arc fitting is "simplifying". The issue with this value is that having a large resolution increases the likelihood of arcs, but it also decreases the quantization of lines. I am not sure having a single value tied to both of these actions is a good idea as the outcome of "more arcs" is "crappier lines". It also seems like every with "comically" large values of resolution, often only certain types of lines are correctly turned into an arc, for example my outer walls are arcs, but my inner walls are lines.

closed 2023-11-20 by github-actions[bot]