OrcaSlicer/OrcaSlicer #629: Internal solid infill is extremely slow

zviratko, 2023-04-03:
**Describe the bug**
When objects are sliced with internal solid infill then the print gets very slow, almost like slow ironing.
This makes the print both extremely slow but also makes the time estimate very inaccurate.

**To Reproduce**
Slice something with realtively thin walls but don't use enough perimeters for them? You can see it in the screenshot.
This is a relatively long print and it's already 3 hours overdue. Looks like increasing perimeters avoids it.

**Expected behavior**
It shouldn't be dreadfully slow and it should at least calculate time properly.

**Screenshots**
[image]
[image]

**Printer model**
X1C

Let me know if you want a 3MF file, I don't have one for the current print exactly as I had to recreate it and it's large, but I can see it in smaller prints also and will reproduce if needed.

SoftFever, 2023-04-03:
I'm unclear about what the issue is here.
You don't want the solid infill? or the printing speed is slow?(which is not at least from your screenshot)
Yes, a project file(.3mf) that can reproduce the bug is the most helpful thing :)

zviratko, 2023-04-03:
Ok, I will provide a small 3mf later.

Yes, the problem is that the solid infill is inexplicably slow. Maybe it doesn't look slow from the gcode, but is slow in reality, which would explain why a ~10 hour print turned into a 14 hour one in this case. My hypothesis would be that it's because of small jerk values and lots of very small movements. In practice it almost looks like the extruder is standing still, which is why I compared it to very slow ironing. Maybe I'll also provide a video so that you can see what it actually does.

Thanks

Neiizo, 2023-04-03:
Hi, i have the same issue. The slicer is saying it is printing the infills at 210mm/s, but in reality, , it prints at 100mm/s AT MOST with very slow acceleration, so that means for small lines, it will go at 10-20mm/s at most. This is reducing the print time by a long margin

zviratko, 2023-04-15:
Sorry for the delay, was waiting for this issue to show clearly on a smaller print and I think I have it
[moai.zip](https://github.com/SoftFever/OrcaSlicer/files/11240671/moai.zip)

Look at layer 116 on the shoulders, it just sits there and barely moves and it takes forever, same happens on the chin I think (and it's not overhangs)
Video showing the issue here: https://www.youtube.com/shorts/1oSgV233yHw (terribly sorry that Youtube made it "short", nothing I can do about that now :( )

closed 2023-11-21 by github-actions[bot]