OrcaSlicer/OrcaSlicer #631: Separate Accel/Decel pressure advance.

EFlynn7778, 2023-04-03:
Issue:  Pressure advance effects not equal with acceleration and deceleration.  Using a single pressure advance setting for both acceleration and deceleration usually equates to only one end being ideal, and the other leaving a blob or gap rather than an ideal extrusion along the entire length of the move.

**Describe the solution you'd like**
2 pressure advance settings for start and end of move in the tuner, as well as print settings to allow fine tuning both ends of the move.

Context:  I can usually only ever get one end perfect leaving the other a small blob. I can move it to either end of the move by changing the pressure advance but can never quite get it ideal for both ends.

Quick-Flash, 2023-04-03:
I also had this thought. When manually tuning my pressure advance I would notice that half would be perfect, and the other half almost perfect. Not sure in practice how much better the prints will end up looking though.

EFlynn7778, 2023-04-06:
> I also had this thought. When manually tuning my pressure advance I would notice that half would be perfect, and the other half almost perfect. Not sure in practice how much better the prints will end up looking though.

I think it would be a decent improvement. 

Its apparent on most prints that have long print lines.  Its usually quite apparent on the first layer, often leaving the lacking end disjoint from the wall a tiny bit, where the other end is perfect.  I have confirmed this is indeed the pressure advance that causes this but moving the setting around. I usually run it in the middle of the decent range so both ends are at least equal. With them separated, we could get both ends perfect.

csdstudio78, 2023-04-06:
This would need to be a klipper modification, not a slicer feature. You can't change PA mid-extrusion because it's already been precalculated by klipper. Try posting in the klipper discord or github. I've always wanted this added to klipper but they say it's not possible or something.

kylek29, 2023-04-06:
> This would need to be a klipper modification, not a slicer feature. You can't change PA mid-extrusion because it's already been precalculated by klipper. Try posting in the klipper discord or github. I've always wanted this added to klipper but they say it's not possible or something.

Might be something someone can manually test with a little GCode by inserting changes to the PA value on the acceleration sequence and again on the deceleration. The change to the Slicer would likely be doing just that.

Quick-Flash, 2023-04-07:
> Might be something someone can manually test with a little GCode by inserting changes to the PA value on the acceleration sequence and again on the deceleration. The change to the Slicer would [...]

If this is possible, it would also require some extra work for the slicer as it would need to mark certain parts of a line as acceleration and other parts of the line as deceleration. This would at most be a best guess as the printer will not end up perfectly matching what we expect it to do. This would still likely be able to give better performance, but it would likely be better if it was added to klipper and the Bambu line of printers. I don't see why it couldn't be possible to do, but I'm also no expert.

closed 2023-11-21 by github-actions[bot]