OrcaSlicer/OrcaSlicer #632: [FR] Convert from "Cloning" models to "Instances" like PS originally does.

MinnesotaMean, 2023-04-03:
My biggest and only main complaint with Bambu's iteration of Prusaslicer is that they opted to remove using instances and instead clone the parts. This is super obnoxious when making changes to one part, as it doesn't propagate to all clones. 
You need to delete all clones and re-clone the original. There also is no good way to set the exact number of parts on the plate, or to fill the plate completely with parts (great for overnight printing)

Is this a possibility to change back to the instance login that Prusaslicer uses?

dvomang, 2023-07-01:
I agree with this
I also wish you could do the fill plate option

programbo, 2023-07-01:
Given the number of times I've cloned and organised parts on the plate before realising I didn't select the desired the filament on the original part, instances would be greatly appreciated.

locki-cz, 2023-09-15:
please implement this

ksipple, 2023-10-31:
I would also like the option to set number of instances of an object without cloning/copying

locki-cz, 2023-10-31:
> Given the number of times I've cloned and organised parts on the plate before realising I didn't select the desired the filament on the original part, instances would be greatly appreciated.

or when you draw seam or make different modification ... you need clone it again ... and again :(

locki-cz, 2024-01-30:
no any plans to ipmelent it ?

CCS86, 2024-03-15:
@SoftFever 

Any chance of making this happen? 

It is so tedious to spend a ton of time meticulously nesting clones on the build plate to be aligned just right, and then realize you want to change a painted seam location, painted support, etc.

CCS86, 2024-06-14:
Pretty please!

locki-cz, 2024-06-15:
When a developer needs it, it will be implemented; otherwise, we'll never get this very basic feature :(

vovodroid, 2024-07-24:
@locki-cz you got it now)))

karcaw, 2025-01-22:
Well hopefully we can unstale it!

locki-cz, 2025-04-23:
@Noisyfox please can you look on this and unstale it?

danvie4, 2025-06-21:
I would really like this feature back as well, it's unfortunate that a slicer that is better in many ways lost some functionality over the original.

closed 2025-10-26 by SoftFever