OrcaSlicer/OrcaSlicer #635: Black on grey is almost impossible to see

kubark42, 2023-04-03:
**Describe the bug**
The color of imported parts is black. This sets up a situation with very low contrast and as a result it is very difficult to use OrcaSlicer.

**Expected behavior**
The default parts color should be high contrast. 

There could also be a means to change the part color, but the default should still be something with high contrast.

**Screenshots**
[image]

SoftFever, 2023-04-03:
Next version will assign a better default color c7dbf848a62b56ea83fa5cc169b961f31d42b979

For now, you can manually change the color by clicking the :
[image]

kubark42, 2023-04-03:
Wonderful!

closed 2023-04-03 by kubark42