OrcaSlicer/OrcaSlicer #636: Bumps on 10-20 degree gentle slopes

adrianke77, 2023-04-04:
**Describe the bug**
I was getting bumps on slopes on prints about 10-20 degrees from horizontal, on default 0.08m print settings.

**To Reproduce**
Print using default settings on a slope about 10 degrees from horizontal, with 15% grid infill.

**Expected behavior**
The buggy behavior is having bumps on slopes on prints about 10-20 degrees from horizontal, on default 0.08m print settings.

The bumps are due to printing the internal solid infill over the sparse infill (this occurs when printing the internals underneath a gentle slope, and the solid infill edge on every layer is being printed over the edge of the sparse area) and the edge of the solid infill was going 350mm/s over the sparse infill and producing lumps as at this speed the bridging was tattered and broken up.

**Screenshots**
[image]

The above photo shows the same 10 degree slope with varying internal solid infill speeds from 175 to 275 mm/s. A workaround to avoid the bumps is to set internal solid infill speed to 200mm/s or below, but IMHO the proper fix would be for the slice to recognise the edge of the solid infill over the sparse infill is a bridging situation and should use the bridging speed variable instead. 

**Printer model**
Bambu X1C

**Desktop (please complete the following information):**
Windows 10

igiannakas, 2023-11-23:
It’s because slanted surfaces use the ensure vertical thickness algorithm which uses the infill speed to extrude. Sounds like a check is needed there to adjust speed depending on whether the area is unsupported? Or the vertical thickness algorithm should use a different speed setting?

@Noisyfox

closed 2024-03-10 by github-actions[bot]

igiannakas, 2024-03-10:
Btw this has been fixed with the introduction of the internal bridge filtering options.