OrcaSlicer/OrcaSlicer #640: Pressure Advance Pattern Method instead of line method

Sure33, 2023-04-04:
**Is your feature request related to a problem? Please describe.**

I've been reading the Ellis Print Tuning Guide and it seems that the line method used to calibrate the pressure advance is no longer a good method. It seems that it can cause inconsistencies.

**Describe the solution you'd like**

The new Pattern Method is far superior to dial the correct PA settings. Described here: [Pattern Method](https://ellis3dp.com/Print-Tuning-Guide/articles/pressure_linear_advance/pattern_method.html)
 It would be great that the Pattern Method could be implemented in the Calibration menu of Orca Slicer.

**Describe alternatives you've considered**
Generate the gcode using the Ellis tool here: [Pressure Advance Tool](https://ellis3dp.com/Pressure_Linear_Advance_Tool/)

Padawg, 2023-04-05:
I saw this also. It would be a great addition to the ORCA slicer. Hopefully it will be implemented in the future.

SoftFever, 2023-04-05:
I'm unsure if adding another calibration method for the same thing is a good idea.(We already have two)
With that said, I'm open if there are other contributors interested in port that method over.

atl285, 2023-04-05:
The advantage I also see is the easier removing from the heatbed. My last pressure advanced test with PETG filament was hard to remove from the Textured PEI plate of my printer. I had the little numbers fly around for days nearby the printer.  ;)

Padawg, 2023-04-05:
Yes I agree, removing the line test is messy. I too am still finding little numbers all over my room. 
I think the more options the better and this test seems to be a newer and easier way to do PA. Sometimes those lines are a pain to read.

EFlynn7778, 2023-04-06:
I dont disagree with the pattern method being better as it is, but a more pressing addition would be to have separate pressure advance settings for the start and end of the moves. This would eliminate the problems with the line method entirely and make the pattern method unnecessary other than for those that prefer it.

kylek29, 2023-04-06:
There's another issue that suggestions as well. I think it's worthwhile, just for aiding in removing the pattern from the bed .. the PA line method is so annoying to remove and the other one uses a lot of filament.

H4kunaM4tata, 2023-04-09:
This would be so awesome, the pattern method allows you to truly dial your PA and so much easier to remove. 
The line is a pain to remove.
But only if OrcaSlicer imports the whole process, not half.

I ran it on my Voron as Elli suggests, run the first time with the default `0.005` increment to find the best value, let's say `0.03`
Then repeat it again between `0.02` to` 0.04` with `0.001` increment

You will find out that the `0.03` you thought was perfect, it is not with the second run.
It might be `0.02xx` or `0.03xx` so why the second run is a must do.

bdistin, 2023-04-15:
I think this is a great idea. As an addendum though, the Calibration menu should be re-ordered for better step-by-step calibration. IE You tune PA before Flow, etc.

H4kunaM4tata, 2023-04-19:
@nirurin , I agree 100%

@SoftFever , the tower should not be considered a method, it has so much room for user error and it is not precise while the lines are just a whole level of pain.

For CoreXY those aren't precise either, I have been using the legend Ellis on my Voron: https://ellis3dp.com/Pressure_Linear_Advance_Tool/

I do as he recommends it:
Find the best result.

[image]

Let's say the best result was `0.03`, run it again from one before and after with `0.001` increment

[image]

You have the gcodes like below.
Notice how everything is connected so easier to remove from the sheet, and for printers that are past 200mm/s past 10k accel, this designed better support PA calibration.
The old line method is not suitable for CoreXY

Thank you for considering this.

[image]

Gil80, 2023-05-10:
Please add this option in the slicer.

claudermilk, 2023-05-11:
Adding to the chorus. I've used the new chevron pattern for both my Trident and Mini+ with great success. Ditch the old lines method and replace with the new one.

H4kunaM4tata, 2023-05-11:
> Adding to the chorus. I've used the new chevron pattern for both my Trident and Mini+ with great success. Ditch the old lines method and replace with the new one.

Trident gang 🙌

thewildmage, 2023-05-26:
> I'm open if there are other contributors interested in port that method over.

I'm interested in giving this a go. @SoftFever, could I please have this Issue assigned to me?

SoftFever, 2023-05-27:
> > I'm open if there are other contributors interested in port that method over.
> 
> I'm interested in giving this a go. @SoftFever, could I please have this Issue assigned to me?

Cool!
Done

SoftFever, 2023-05-27:
> > I'm unsure if adding another calibration method for the same thing is a good idea.(We already have two) With that said, I'm open if there are other contributors interested in port that method over.
> 
> You already have two, but one of them is inaccurate to use and the other is a pain in the neck to clean up. Remove them both and replace them with the new version that is both easier and faster to use, and simple to clean up.
> 
> Imo it's a no brainer. The existing methods have zero reasons to exist.

That's your personal opinion. 
Let's agree to disagree.

nirurin, 2023-05-29:
> > I'm open if there are other contributors interested in port that method over.
> 
> I'm interested in giving this a go. @SoftFever, could I please have this Issue assigned to me?

Good job taking this on, I'm sure your work will be appreciated! Lots of people on the bambu discord have been waiting for this change, noone likes using the old methods. If I could tip you I would!

dvomang, 2023-06-21:
Any updates on this?

R4toLoko, 2023-06-21:
> Any updates on this?

I don't think this will happen anything soon.

He said it loud and clear that there are already 2 methods in place which we all know they shouldn't be considered as such, the point is, there are 2 methods already so there is no reason to add another one.
When you reality, he just need to remove those 2 "useless method" in flavour for Ellis' method.

A legend volunteered to make it available, that is the only hope we have.
As of now, is to keep accessing Ellis' wiki.

thewildmage, 2023-06-21:
> Any updates on this?

I'm definitely still working on it! It's turned out to be trickier than I expected, but I'm making progress and am determined to see it through. Stay tuned!

PR #1547 (2023-07-22): Add Ellis' pattern method for pressure advance calibration
~**WIP note:** I'm feeling good about where the code for this is at, ready to see how it does with the build checks, but I also want to do some actual print tests before moving it out of WIP status. Will update once those are done.~
Done! Bugs and optimization opportunities identified and incorporated.

---

Closes #640. Closes #794.
Also heads up to #373 and #761.

I was going to pull the instructions into `doc/Calibration.md`, but ran into a licensing issue with that. The [pattern generator](https://github.com/AndrewEllis93/Pressure_ [... cut]
closed 2023-07-22 by SoftFever