OrcaSlicer/OrcaSlicer #644: After slice, not printer dialogue is shown to print the plate

grindfreak1981, 2023-04-05:
So, i have been enjoying OrcaSlicer since it's official release. Last week, it suddenly stopped working as intended. I can slice files, play around with settings, but when i want to print the plate, nothing happens. The only option i have, is to reslice the plate, and still can't send the file to printer. The dialogue doesn't pop up. Also, when i want to save any settings or projects, it crashes. If this already exists somewhere, i apologize, but i couldn't find any threads regarding this issue.

hliebscher, 2023-04-05:
witch operating system and witch OrcaSlicer version?

grindfreak1981, 2023-04-05:
It’s windows 11 64-bit, any version of orca slicer. I mean there is only
1.5.0. I had first orca slicer and today i tried to redownload it, and
still nothing. Last softfever works just fine.

closed 2023-11-21 by github-actions[bot]