OrcaSlicer/OrcaSlicer #650: Sliced Gcode sent to Mainsail "KLIPPER" will not start print after update

Tyler8299, 2023-04-05:
Console LOG: starts with Malformed command. Thank You. v1.5.0 still slices normal for now ill use that for my enders

Extruder not hot enough
3:14 PM
Extruder not hot enough
3:14 PM
Printer not homed
3:14 PM
Malformed command 'EXCLUDE_OBJECT_DEFINE NAME=Ultimate Box Snap Latch_zx82net rugged box 2a v2.stl_id_0_copy_0 CENTER=81.6726,97.3902 POLYGON=[[61.123,102.217],[61.1286,102.117],[61.415,100.079],[61.4628,99.7597],[61.5152,99.4487],[61.5722,99.1458],[61.6338,98.8508],[61.6998,98.5636],[61.7703,98.2843],[61.8452,98.0125],[61.9245,97.7484],[62.008,97.4918],[62.0958,97.2426],[62.1877,97.0008],[62.2838,96.7662],[62.3839,96.5389],[62.488,96.3185],[62.5959,96.1049],[62.7077,95.8981],[62.8231,95.6979],[62.9421,95.5041],[63.0646,95.3167],[63.1905,95.1354],[63.3198,94.9603],[63.4523,94.791],[63.5879,94.6276],[63.7267,94.4699],[63.8684,94.3178],[64.0131,94.1712],[64.1606,94.03],[64.3108,93.8941],[64.4638,93.7634],[64.6193,93.6378],[64.7774,93.5171],[64.938,93.4014],[65.1009,93.2905],[65.2662,93.1843],[65.4337,93.0828],[65.6035,92.986],[65.7754,92.8938],[65.9495,92.8062],[66.1257,92.7231],[66.3039,92.6445],[66.4841,92.5703],[66.6663,92.5006],[66.8503,92.4352],[67.0362,92.3742],[67.2239,92.3174],[67.4134,92.2651],[67.6047,92.217],[67.7978,92.1734],[67.9926,92.1342],[68.1892,92.0993],[68.3875,92.0689],[68.5876,92.043],[68.7894,92.0215],[68.993,92.0045],[69.1982,91.992],[69.4052,91.984],[69.6139,91.9806],[69.8243,91.9816],[70.0364,91.9873],[70.2501,91.9975],[70.4655,92.0124],[70.6826,92.0319],[70.9014,92.056],[71.1218,92.0847],[97.9149,95.8494],[98.233,95.9029],[98.5478,95.9733],[98.8584,96.0606],[99.1638,96.1645],[99.4632,96.2846],[99.7558,96.4207],[100.041,96.5723],[100.317,96.7389],[100.584,96.9202],[100.84,97.1156],[101.086,97.3244],[101.321,97.5462],[101.543,97.7803],[101.752,98.0259],[101.947,98.2825],[102.129,98.5492],[102.175,98.6365],[102.207,98.7302],[102.222,98.8278],[102.222,98.9267],[102.205,99.0241],[102.172,99.1174],[102.125,99.2041],[102.063,99.2818],[101.99,99.3484],[101.907,99.4021],[98.5972,101.177],[98.5104,101.215],[98.4188,101.239],[98.3245,101.248],[61.7001,102.8],[61.6006,102.788],[61.5045,102.759],[61.4145,102.715],[61.3332,102.657],[61.2626,102.585],[61.205,102.503],[61.1617,102.413],[61.1342,102.317],[61.123,102.217]]'

Tyler8299, 2023-04-05:
seems Klipper does not understand the added code incorporated with new sliced gcode dealing with exclude object start and end lines.

Tyler8299, 2023-04-05:
[image]
[image]

Tyler8299, 2023-04-05:
[image]

Styxbleich, 2023-04-05:
[image]

I have his same issue as well.

Tyler8299, 2023-04-05:
Uncheck Exclude Object and leave label object checked klipper knows what to do with the labeled Object. seems to be working so far

Tyler8299, 2023-04-05:
UPDATE MOONRAKER!!!! im deleting this

TexZeTech, 2023-04-06:
> Uncheck Exclude Object and leave label object checked klipper knows what to do with the labeled Object. seems to be working so far

https://www.klipper3d.org/Exclude_Object.html

https://www.obico.io/blog/klipper-exclude-object/

SoftFever, 2023-04-06:
Thanks for reporting back.
I think the issue is caused by the spaces in the model name.
I will release a hotfix for it.

screech24, 2023-04-06:
> UPDATE MOONRAKER!!!! im deleting this

can you confirm which moonraker version you are on thats working?  Because I am on the latest available to me .. v0.8.0-32-g31e589a
 and the only way to make it work is unchecking Exclude objects...

Tyler8299, 2023-04-06:
Version Moontraker v0.8.0-32-g31e589a

Tyler8299, 2023-04-06:
> > UPDATE MOONRAKER!!!! im deleting this
> 
> can you confirm which moonraker version you are on thats working? Because I am on the latest available to me .. v0.8.0-32-g31e589a and the only way to make it work is unchecking Exclude objects...

Moontraker v0.8.0-32-g31e589a

screech24, 2023-04-06:
> Thanks for reporting back. I think the issue is caused by the spaces in the model name. I will release a hotfix for it.

FYI removing the spaces doesn't fix the issue, for whatever reason the spaces get added back once sliced and sent to the printer. Also unchecking the exclude objects box allows the print to work but the feature is disabled if you do so in my testing anyway. THE WAY TO FIX THE ISSUE IS TO INSURE YOU ARE ON THE LATEST VERSION OF MOONRAKER AND RESTART THE PI. ( restarting Moonraker on its own did not work for me.)

screech24, 2023-04-06:
> > > UPDATE MOONRAKER!!!! im deleting this
> > 
> > 
> > can you confirm which moonraker version you are on thats working? Because I am on the latest available to me .. v0.8.0-32-g31e589a and the only way to make it work is unchecking Exclude objects...
> 
> Moontraker v0.8.0-32-g31e589a

Indeed upating moonraker and rebooting the pi fixed the issue for me. This issue should be marked as closed.

AMFkuna, 2023-04-06:
Does not work for me, updated moonraker, restarted PI - not works.

Tyler8299, 2023-04-06:
> It's a legit bug. When the part name contains spaces, it will trigger it.

Thank You for your hard work. as always its greatly appreciated.

SoftFever, 2023-04-06:
Fixed in V1.6.1

closed 2023-04-06 by SoftFever