OrcaSlicer/OrcaSlicer #651: User Filament Preset Don't Have "None" For Filament Color

eboston, 2023-04-05:
**Describe the bug**
In previous version, when a user filament preset was created, the filament color was "none" so that the color set in the AMS would be used.  After upgrading to v1.6.0, all y preset have changed to a set color and I can't set them back to "none" to get the color from the AMS.

**To Reproduce**
Upgrade to v1.6.0 and use an existing user filament preset.

**Expected behavior**
The color is not defined in the preset and the AMS color is used.

**Desktop (please complete the following information):**
 - OS: Windows 10

If a user filament preset explicitly defines a color, use that, otherwise use the color defined in the AMS

Photogad, 2023-04-10:
+1

This is a major inconvenience, needs to be fixed.

Photogad, 2023-04-13:
Not only is it a visual problem, but it seems this bug also translates to the printing window -- even when selecting a custom filament from the list in the slicing process, it does not correctly choose that AMS filament. I've had twice now the AMS choose the wrong filament and printed wrong color.

@SoftFever Can you confirm if this is an OrcaSlicer bug or also present in main Bambu Studio fork?

SoftFever, 2023-04-18:
Fixed in 1.6.2 beta

closed 2023-04-18 by SoftFever