OrcaSlicer/OrcaSlicer #652: Orca Slicer V1.6 Linux app image segmentation fail

A-nglea, 2023-04-05:
Started completely fresh. 
Removed all .local/share/orca-slicer , .config/OrcaSlicer and .cache/orca-slicer/
 ( one wonders at naming conventions with OrcaSlicer being the odd one out)  
Ran the app image on Fedora. (App images should run anywhere - right?)
Managed to get to login
and then this. Repeatable.In fact it has been repeatable since forever.
 

< Date: Wed, 05 Apr 2023 20:26:43 GMT
< Content-Type: application/json
< Content-Length: 90
< Connection: keep-alive
< Server: APISIX/2.15.0
< X-Request-Id: 1be926ce-3cd6-4589-a299-ebd7d0ff454f
< 
* Connection #0 to host api.bambulab.com left intact
Segmentation fault (core dumped)

And from dmesg

[22618.737388] bambustu_main[3908]: segfault at 50 ip 0000000002f36d88 sp 00007feb89ff1530 error 4 in bambu-studio[4a5000+3144000] likely on CPU 1 (core 1, socket 0)
[22618.737426] Code: 00 00 b8 01 00 00 00 48 81 c4 58 01 00 00 5b 5d 41 5c 41 5d 41 5e 41 5f c3 66 0f 1f 84 00 00 00 00 00 48 89 df e8 08 f9 56 fd <48> 39 68 50 75 1a 8b 50 40 83 fa 02 0f 84 f6 09 00 00 83 fa 40 0f
[38606.008543] bambustu_main[41264]: segfault at fffffffffffffffa ip 00007fbe948be30e sp 00007fbc8d3f4b40 error 5 in libc.so.6[7fbe94849000+156000] likely on CPU 0 (core 0, socket 0)
[38606.008573] Code: 08 5b fe ff 0f 1f 84 00 00 00 00 00 f3 0f 1e fa 48 85 ff 0f 84 bb 00 00 00 55 48 8d 77 f0 53 48 83 ec 18 48 8b 1d 02 7b 13 00 <48> 8b 47 f8 64 8b 2b a8 02 75 57 48 8b 15 88 7a 13 00 64 48 83 3a
[38622.011264] bambustu_main[41929]: segfault at fffffffffffffffa ip 00007f99efabe30e sp 00007f98f5ff4b40 error 5 in libc.so.6[7f99efa49000+156000] likely on CPU 6 (core 2, socket 0)
[38622.011295] Code: 08 5b fe ff 0f 1f 84 00 00 00 00 00 f3 0f 1e fa 48 85 ff 0f 84 bb 00 00 00 55 48 8d 77 f0 53 48 83 ec 18 48 8b 1d 02 7b 13 00 <48> 8b 47 f8 64 8b 2b a8 02 75 57 48 8b 15 88 7a 13 00 64 48 83 3a
[38682.585454] bambustu_main[42600]: segfault at fffffffffffffffa ip 00007f34766be30e sp 00007f1f95f2db40 error 5 in libc.so.6[7f3476649000+156000] likely on CPU 7 (core 3, socket 0)
[38682.585468] Code: 08 5b fe ff 0f 1f 84 00 00 00 00 00 f3 0f 1e fa 48 85 ff 0f 84 bb 00 00 00 55 48 8d 77 f0 53 48 83 ec 18 48 8b 1d 02 7b 13 00 <48> 8b 47 f8 64 8b 2b a8 02 75 57 48 8b 15 88 7a 13 00 64 48 83 3a
[38821.998130] perf: interrupt took too long (3994 > 3925), lowering kernel.perf_event_max_sample_rate to 50000
[43551.926552] bambustu_main[50318]: segfault at fffffffffffffffa ip 00007fa93a2be30e sp 00007fa8423f5b40 error 5 in libc.so.6[7fa93a249000+156000] likely on CPU 5 (core 1, socket 0)
[43551.926580] Code: 08 5b fe ff 0f 1f 84 00 00 00 00 00 f3 0f 1e fa 48 85 ff 0f 84 bb 00 00 00 55 48 8d 77 f0 53 48 83 ec 18 48 8b 1d 02 7b 13 00 <48> 8b 47 f8 64 8b 2b a8 02 75 57 48 8b 15 88 7a 13 00 64 48 83 3a
[43854.522183] bambustu_main[51446]: segfault at fffffffffffffffa ip 00007f7bc8ebe30e sp 00007f79d53f5b40 error 5 in libc.so.6[7f7bc8e49000+156000] likely on CPU 1 (core 1, socket 0)
[43854.522217] Code: 08 5b fe ff 0f 1f 84 00 00 00 00 00 f3 0f 1e fa 48 85 ff 0f 84 bb 00 00 00 55 48 8d 77 f0 53 48 83 ec 18 48 8b 1d 02 7b 13 00 <48> 8b 47 f8 64 8b 2b a8 02 75 57 48 8b 15 88 7a 13 00 64 48 83 3a

bojangles86, 2023-04-05:
is this the login issue where it freezes for a moment before crashing? because i can use orca as long as i dont login which kinda defeats a lot of the built in wifi functions

SoftFever, 2023-04-06:
This is a known issue on Fedora.
I personally don't use Fedora so I don't have a solution for that.

savageawakening, 2023-07-13:
Just got the same issue, also on Fedora but not during login, rather I was trying to connect to a LAN mode P1P and it crashed. Same segfault in libc.so.6. I am not familiar with packaging for appimages but would it be possible to include glibc dependencies so it uses those instead of the system ones?

ikilledmypc, 2023-07-19:
I see that the Bambu release has a specific binary for fedora, I will try to find the difference when I have the time and see if I can make a PR.

ikilledmypc, 2023-07-21:
By following the instructions for building on the Bambu Lab studio on Fedora, I managed to build Orca slicer under fedora from source. It does have the upstream issue with a dismiss able js error popup and the camera does not work. I have no idea how the appimage is created for fedora for bambu lab studio but at least I can get it to slice and print now

A-nglea, 2023-07-21:
You might want to check back on Bambu Lab Studio the current version release notes say:-
"This is a major release (V1.7.0 Public Beta) mainly adds some features, improvements, and fixes based on user feedback.
This version has incorporated many features from [OrcaSlicer](https://github.com/SoftFever/OrcaSlicer)"

closed 2023-11-21 by github-actions[bot]