OrcaSlicer/OrcaSlicer #654: Keyboard issues - Linux AppImage

rmcfadzean, 2023-04-05:
**Describe the bug**
Issue 1:
After clicking the search icon to search configuration options, the search box appears, but no key presses are registered and the search isn't narrowed.

Issue 2:
`backspace` (as noted by the context menu when opened on an object) does not delete the object from the build plate. Neither does `delete`.
All other hotkeys seem to work as expected.

**To Reproduce**
Issue 1:
* Use the AppImage on Arch Linux
* Click the search icon next to the profile selection dropdown
* Try to enter a search term
* Be sad that it doesn't work

Issue 2:
* Add an object to the build plate
* Click on the object to select it
* Press `backspace` on your keyboard to delete it
* Be sad again
* Try pressing `delete` just in case
* Continue sadness and use the context menu to delete the object

**Expected behavior**

Issue 1:
Be able to type into the search box and narrow search results

Issue 2:
Delete parts from the plate using a hot key

**Screenshots**
[image]

**Printer model**
N/A

**Desktop (please complete the following information):**
 - OS: Arch Linux - Kernel: 6.2.9-arch1-1
 - Version v1.6.0  (goes back to at least 1.4.0)
 - Package: Official .AppImage release from GitHub

TexZeTech, 2023-04-29:
Version: 1.6.2 stable

[image]

**System & Hardware:**
Operating System: Manjaro Linux 
KDE Plasma Version: 5.27.4
KDE Frameworks Version: 5.104.0
Qt Version: 5.15.8
Kernel Version: 6.1.25-1-MANJARO (64-bit)
Graphics Platform: X11
Processors: 4 × Intel® Core™ i5-3337U CPU @ 1.80GHz
Memory: 3.3 GiB of RAM
Graphics Processor: Mesa Intel® HD Graphics 4000
Manufacturer: ASUSTeK COMPUTER INC.
Product Name: TAICHI31
System Version: 1.0

TexZeTech, 2023-07-14:
This seems to be working on my system as of OrcaSlicer V1.6.3 in both the appimage and AUR image @rmcfadzean.

theotherjimmy, 2023-08-09:
I don't have issue 1, but I do have issue 2

TexZeTech, 2023-08-09:
I just realized I only confirmed the search, not the delete button. :facepalm:

TexZeTech, 2023-09-12:
So as of 1.7.0 beta appimage (possibly a few images ago) the delete key works for removing parts on the build plate & search works as expected.

```
Operating System: Manjaro Linux 
KDE Plasma Version: 5.27.7
KDE Frameworks Version: 5.109.0
Qt Version: 5.15.10
Kernel Version: 6.1.51-1-MANJARO (64-bit)
Graphics Platform: X11
Processors: 8 × Intel® Core™ i7-3770K CPU @ 3.50GHz
Memory: 15.6 GiB of RAM
Graphics Processor: NVIDIA GeForce GTX 1080/PCIe/SSE2
```