OrcaSlicer/OrcaSlicer #662: "reload from disk" shortcut

FillPavS, 2023-04-06:
Hello! I just switched from superslicer. And I did not find that the F5 hotkey responsible for updating the model on the table (reboot from disk) does not work. I very often make changes to the model and then watch how it is sliced to select the best result or when you just need to increase the gap. Now, to update the model, you need to switch the tab from global to objects, right-click on the object and select update, or in the preparation window, right-click on the model and select this item. Please make it possible to update with one button. It was very convenient to press F5 and Ctrl+R right in the preview window to instantly get an update

[image]

closed 2023-11-21 by github-actions[bot]

radicalgale, 2024-05-31:
This is a very good idea. I use "Reload from disk" feature a lot. It shouldn't be that hard to update the model. Cura detects changes automatically, SuperSlicer allows updating the mesh by pressing F5. It is very useful for rapid prototyping.

jbjhjm, 2024-06-21:
With the objects explorer view being hidden within Process > objects, such a shortcut becomes even more important. I too think it should be reintroduced.