OrcaSlicer/OrcaSlicer #667: Post-processing is being processed twice

H4kunaM4tata, 2023-04-06:
I have klipper_estimator like so: `klipper_estimator --config_moonraker_url http://192.168.1.36 post-process`

And a script to fetch the new time and add to the file like so: ` ./time.sh fan_0.32mm_PLA_TIME_170g.gcode`

But the name gets all weird.

[image]

As you can see, it is being processed twice
The others were sliced with PS and they are just fine.

[image]

With PS it works just fine. Delete the file, slice it and run the script.
One change only, not twice.

[image]

H4kunaM4tata, 2023-04-06:
The work around to avoid manual stuff is to filter only the first line like so: `cat fan_0.32mm_PLA_TIME_170g.gcode | grep normal | awk '{print $8 $9}' | head -n 1`

SoftFever, 2023-04-06:
What does it look like without using klipper_estimator ?

H4kunaM4tata, 2023-04-06:
> What does it look like without using klipper_estimator ?

The file name as usual: `fan_0.32mm_PLA_TIME_170g.gcode`

The script does nothing more than check the gcode for the time that klipper_estimator adds to the file and append it to the filename.

H4kunaM4tata, 2023-04-10:
> What does it look like without using klipper_estimator ?

@SoftFever I think that I didn't reply to your question properly.

With or without having the klipper estimator, OrcaSlicer GCODE file have both lines:

[image]

[image]

PrusaSlicer GCODE it is only present once

[image]

Fix: remove the extra time from the start file, that is the extra piece of information and it is not needed. 

**For what matters, ignore the time difference. Totally not relevant here.**

SoftFever, 2023-04-10:
Thanks for reporting it.
The estimation comment is written twice. It will be fixed in the update.

H4kunaM4tata, 2023-04-10:
Awesome, thank you for checking this.

closed 2023-04-10 by H4kunaM4tata