OrcaSlicer/OrcaSlicer #668: Klipper Adaptive Meshing does not work with the latest version

VDechev, 2023-04-06:
I was using the slicer fine on version V1.4.5. Today I updated to the latest version 1.6.1 and faced issue with Klipper Adaptive Meshing. 
When files are sliced and forwarded to klipper for printing I'm seeing this error when trying to perform KMP "bed_mesh: min/max points too close together"

I suppose with the latest exclude object features, KMP is unable to determinate which is the printing area for the project.

To reproduce this issue you can use bunch of files sliced and sent to the printer.

What we expect is to bed mesh only the printed area

I'm using Voron 2.4 in this case. 
[image]

I have updated moonraker to the latest version

I'm slicing on Windows

TexZeTech, 2023-04-07:
Try V1.5.0 as I'm having an issue (not related to this one)  in V1.6.1

VDechev, 2023-04-07:
> Try V1.5.0 as I'm having an issue (not related to this one) in V1.6.1

In V1.5.0 I'm facing issue with the position of the plates. I have around hundred project with multiple plates which I have to arrange again in V1.5.0. I went back to 1.4.5 where everything is fine

sabaatworld, 2023-04-09:
I'm also facing the same issue with multiple plates

piotrb, 2023-04-11:
I'm testing this now, but it seems like it may be related to the "Exclude Objects" option that was added in 1.6 .. it generates the EXCLUDE_OBJECTS code (incorrectly) .. if you turn that off you go back to 1.5 behavior of relying on Moonraker to generate that .. and it seems to work better maybe .. 

(I've noticed that it generates it with the Center of the object being on negative Y .. so this may be some bug with using the wrong coordinate system .. absolute vs plate specific .. I'd assume this works fine on the first plate)

cc @SoftFever

piotrb, 2023-04-11:
yep with that option turned off and no other changes the KAMP macro works like a charm and the output in the gcode looks more sane with the center being actually on the build plate

piotrb, 2023-04-11:
(the built in support would be ostensively better since it does  a more complex polygon of your object so the meshing can get the area better .. but this is no worse than 1.5 at least

VDechev, 2023-04-11:
Hey @piotrb,

You are talking about turning off the new exclude object which was released in 1.6.1. I would love to go back to 1.6.1 if I can turn off that function. Do you mind sharing how do I do that? Thanks

kyleisah, 2023-04-13:
Hey y'all! I just wanted to comment here that I am following along. I run the KAMP repo and have been getting some issues opened about probe points being too close, because orcaslicer is improperly defining object size variables. The issue is on my repo, issue number 52. I will reference it there. Thank you!

SoftFever, 2023-04-13:
fixed in dev branch

kyleisah, 2023-04-13:
Awesome! Thank you @SoftFever! 😄

piotrb, 2023-04-15:
> Hey @piotrb,
> 
> You are talking about turning off the new exclude object which was released in 1.6.1. I would love to go back to 1.6.1 if I can turn off that function. Do you mind sharing how do I do that? Thanks

In your print settings, under Other, is the new checkbox called "Exclude Objects" turn that off and 1.6.1 will work like 1.5

SoftFever, 2023-04-17:
@kyleisah I tried your KAMP on my Voron 2.4 today, love it!
Well done

coffeewhistle, 2023-04-17:
I might have found the workaround/cause. I was having the same bed mesh min/max issue with KAMP on OrcaSlicer. 

It appears to be related to having multiple build plates visible in the slicer. Even though I'm only slicing and printing one of the plates, if there are other plates visible/set up in the file that I'm trying to print from to my Mainsail/Klipper printer it will throw the error. If I go back to the same exact file, remove the extra build plates from the file and reprint, I have no issue.

SoftFever, 2023-04-17:
Hey guys, this issue has been fixed in 1.6.2 release.

SoftFever, 2023-04-18:
Fixed in 1.6.2 beta

closed 2023-04-18 by SoftFever

kyleisah, 2023-04-20:
@SoftFever You da man! Nice work! 😄