OrcaSlicer/OrcaSlicer #670: Model URL in Project Basic Info

chaugan, 2023-04-06:
**Is your feature request related to a problem? Please describe.**
No problem, just new feature so it's easier to remember where the model was from when saving a project

**Describe the solution you'd like**
Add this new line of info, and a "Go!" link at the end so that the url specified opens in the users default browser.

[image]

SoftFever, 2023-04-07:
I like this idea.
Thanks.

bitflung, 2023-06-15:
We've regressed: now in OrcaSlicer 1.6.3 the Project tab no longer allows for editing any of the expected fields (this matches the broken behavior of Bambu Studio 1.6.2.4)

closed 2023-11-21 by github-actions[bot]