OrcaSlicer/OrcaSlicer #674: Implement a true LAN only mode for Orca

julie777, 2023-04-06:
**Is your feature request related to a problem? Please describe.**

The Bambu Lab networking dll is terrible. It has problems. You can see in the networking log that it is constantly connecting to their servers, even with a LAN only printer being used. The way Bambu Studio was written is very cloud biased and does not maintain the MQTT connection to the printer which causes long delays and timeouts.

One example of this:
1. I am on the device page for my printer and I can see status changing and be viewing the camera feed. Therefore I am connected and the current printer status is known.
2. I shift to preview and "print plate"
3. the print dialog comes up and says it is syncing status of the printer
4. then it times out
5. then I retry and it updates
6. then I send the print and after it has completed the send it jumps to the device page
7. then it has to connect to the printer before it can show that it has started the print process
8. eventually the camera feed starts working.

I have spent many, many hours working through all this with Bambu Studio developers and Bambu Lab support.
They intentionally keep disconnecting and reconnecting because they built it for the cloud and maintaining cloud connections costs them money. For us LAN users this is a terrible implementation.

The Bambu Networking code also constantly talks to bambu servers, looking for new printers and making requests to api.bambulab.com which slows down communication with the printer.

**Describe the solution you'd like**

I would like for Orca to work in a LAN mode if there are only LAN printers connected.
Discovering printers not on the LAN should be a specific command and not something done constantly.
Talking to LAN printers should be done directly, and not use the bambu_networking.dll. There is also an additional closed source dll they load which I haven't determined the purpose for.

**Describe alternatives you've considered**
I have considered building Bambu Studio for myself and implementing the lan only solution, but I think it would be better for the community to have only one better solution (If Bambu Lab is not going to fix their software.)

**Additional context**
Add any other context or screenshots about the feature request here.

WZ9V, 2023-08-26:
Would love a fully functional LAN mode in Linux also.  I would like it to only reach out to Bambu Labs for Firmware and Filament Profile updates if possible.

closed 2023-12-04 by github-actions[bot]

tflucke, 2024-12-07:
I would also like this feature,