OrcaSlicer/OrcaSlicer #675: Can't login with third party auth on mac

alexdibattista, 2023-04-06:
**Describe the bug**
Can't login with third-party auth

**To Reproduce**
Click Login/Register
Click Google under Login with other accounts
nothing happens

**Expected behavior**
Clicking google auth opens a browser window to login

**Printer model**
Bambu X1C

**Desktop (please complete the following information):**
 - OS: Mac 
 - Version: Ventura 13.1

hliebscher, 2023-04-06:
mac m1 it is the same...

and update info not working

Error getting: `https://api.github.com/repos/softfever/OrcaSlicer/releases`: HTTP 0, curl:SSL connect error:
error:1012606B:elliptic curve routines:EC_POINT_set_affine_coordinates:point is not on curve
[Error 35]

alexdibattista, 2023-04-06:
The issue appears upstream in Bambu Studio [issue 434](https://github.com/bambulab/BambuStudio/issues/434). The workaround is installing an earlier Bambu studio version, logging in, and upgrading. This doesn't work OrcaSlicer because Mac complains that versions earlier than 1.5 are corrupted.

alexdibattista, 2023-04-07:
I've found a workaround for OrcaSlicer. You can click forgot password and reset your account to use email and password instead of OAuth.

rosskinard, 2023-05-15:
Getting the same issue with Google login. Just installed 1.6.2 since I ordered the X1 Carbon and AMS today. Login with Google was fine with BambuStudio 1.6.2.4 so curious if this will be fixed soon. Seen a lot of YouTube folks with the Bambu printers that like the features Orca adds in, so thought I'd check it out.

Using a M2 Max MacBook Pro with 13.3.1 (a) (22E772610a)

closed 2023-11-20 by github-actions[bot]