OrcaSlicer/OrcaSlicer #676: Z-seam is present in Spiral Vase mode in OrcaSlicer

sensaurabh, 2023-04-06:
Whenever 'Spiral Vase' mode is selected in the 'Others' tab and an object is sliced, there is a z-seam present instead of smooth transition in the layers.

Can be easily reproduced by selecting a 'Cylinder' primitive from the right click menu and then slicing it after selecting the 'Spiral Vase' checkbox

Expected behavior would be a sliced single walled object with no z-seam

[image]

**Printer model**
KP3S Pro S1
**Desktop (please complete the following information):**
 - OS: [MacOS Ventura]
 - Version [1.6.1]

Notes:

- could be happening because of there is no option to disable the 'Timelapse' mode?
- There is no option to disable the 'z-hop' in the printer menu?
- Have tried changing 'Seam gap' value, but it has had no effect on the seam generation.

hliebscher, 2023-04-06:
Mac M1   OS 1.6.1 it works

sensaurabh, 2023-04-07:
Strange, for me I cannot get rid of the seam or the retractions at the white points you see in the included pictures. It's not just in the preview either.

hliebscher, 2023-04-07:
[image]
[image]

can you send your 3mf  as zip?

sensaurabh, 2023-04-08:
Here it is...  [Cylinder-zSeamInVaseMode.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/11182268/Cylinder-zSeamInVaseMode.3mf.zip)

Does the 3mf also carry over the Filament settings? Do you have retractions enabled in your filament settings?

rcxrbt, 2023-04-12:
Solved?

andrewboktor, 2023-12-12:
I've been slicing in vase mode a lot and never saw this. Must be specific to config.

closed 2023-12-17 by SoftFever