OrcaSlicer/OrcaSlicer #679: Estimated endtime

foxmckraut, 2023-04-06:
Feature Request for Estimated endtime like evey frontend(mainsail/fluid/octoprint) hast.

shows the time when the print is done.

Bonus: shows the dissadvantege betwen sliced time printing time an real printing time.

TexZeTech, 2023-04-07:
It does though?

[image]

shauneccles, 2023-04-07:
I'm guessing they're after:
Total time	22m21s
Estimated endtime	{now + 22m21s}

foxmckraut, 2023-04-07:
> I'm guessing they're after: Total time 22m21s Estimated endtime {now + 22m21s}

^this

it only shows the time it needs not when its done (like 14:42oclock or something) and it didnt show how much more it takes than sliced f.e. 6h sliced takes 8hours in reality.

closed 2023-11-20 by github-actions[bot]