OrcaSlicer/OrcaSlicer #680: Top surface flow also changes line width, in effect cancels out the flow.

Gizmo6, 2023-04-06:
**Describe the bug**
Top surface flow ratio
found in: Quality/advanced/top surface flow ratio. When editing this value for a cleaner top surface it auto adjusts the line width to compensate for the change which in effect cancels out the flow compensation. 

**To Reproduce**
load 2 identical printable files, eg cubes. edit Top surface flow ratio, say 0.5 value and slice. you will see under the preview the line width of the edited one has also changed, but it isn't refected in the top surface line width option. 

**Expected behavior**
when editing top surface flow ratio, I would expect only the amount of material to change but to keep to the line width as specified in the top surface line width option.

**Screenshots**
attached

**Printer model**
bambu lab x1c

**Desktop (please complete the following information):**
 - OS: windows
 - Version 11
[image]

SoftFever, 2023-04-07:
@Gizmo6, can you try the classic wall generator? 
Does it happen on older versions too?

Gizmo6, 2023-04-07:
Thanks for replying. I can confirm that both Arache and classic both display the same behavior.
I can also confirm versions 1.45 and 1.34 also exhibited the same behavior. 

Your calibration tools are fantastic but as the flow samples are so flat for some reason it gives a perfect result. But in order to test my theory I made the 9 flow tests 5mm tall and the top surface was again poor. 

[image]

netweaver1970, 2023-04-07:
I recognize the flat vs real size behavior too. Tuned to the flat pieces perfectly but when printing real life objects, the top level is never that nice.

SoftFever, 2023-04-07:
The top surface top flow is a bug.

For the flow rate calibration, please move to https://github.com/SoftFever/OrcaSlicer/discussions for suggestions/ideas/discussions.

SoftFever, 2023-04-07:
And it's just a gcode preview/rendering bug.
The actual g-code is fine.

Gizmo6, 2023-04-07:
The render is exactly the behaviour of the actual print as you can see from the photos. I have tried to manually play with the adjusting the top surface line width to compensate for the behaviour but its just abit of a mess. 

Just wanted to say absolutely love your work on the slicer. The top surface flow was my main reason to switch in the beginning and i am sure it did work in the original. I am trying to find the original version now to test the behaviour.

SoftFever, 2023-04-07:
> 

yeah, it's a legit bug.

Gizmo6, 2023-04-07:
Amazing job will get it on the printer for testing once first print is done.

closed 2023-04-18 by SoftFever

sylanc3, 2023-04-21:
Hello together, I'm not a developer, but I want to report something:
I use 1.62 beta and the Top Surface flow ratio shows no effect on actual print. Is there maybe another bug?
[image]
left to right
1.0 1.05 1.1 1.2

SA7BNT, 2023-05-31:
Still having that issue that it is not possible to adjust the Top Surface flow ratio.
Running 1.6.3 Mac Intel Version