OrcaSlicer/OrcaSlicer #683: Missing tempreature information from start machine code?

TexZeTech, 2023-04-07:
I have attached the two full gcode files below from the v1.5.0 and v1.6.1

**Describe the bug**
Between OrcaSlicer v1.5.0 and v1.6.1 there is a difference in how my start gcode is being read as its dropping my filament temperature. Looking at the first few lines in the gcode files there is no m190 m104 or m109 

**To Reproduce**
Sliced on 1.5.0 appimage and on 1.6.1 with Exclude objects disabled/enabled

**Expected behavior**
my printer after heating up to the 240°C for my PETG and the bed to 90°C would then move the X axis to the right and the Z down.

What it actually does is move down then to the right now and run at 200°C on the hotend resulting in a blob.

**Printer model**
Robo3D R1 heavily modified running Klipper & Mainsail

**Desktop:**
Operating System: Manjaro Linux 
KDE Plasma Version: 5.27.3
KDE Frameworks Version: 5.104.0
Qt Version: 5.15.8
Kernel Version: 6.1.21-1-MANJARO (64-bit)
Graphics Platform: X11
Processors: 8 × Intel® Core™ i7-3770K CPU @ 3.50GHz
Memory: 15.6 GiB of RAM
Graphics Processor: NVIDIA GeForce GTX 1080/PCIe/SSE2

**Machine Start code**

```
PRINT_START EXTRUDER=[nozzle_temperature_initial_layer] BED=[bed_temperature_initial_layer_single]
; You can use following code instead if your PRINT_START macro support Chamber and print area bedmesh
; PRINT_START EXTRUDER=[nozzle_temperature_initial_layer] BED=[bed_temperature_initial_layer_single] Chamber=[chamber_temperature] PRINT_MIN={first_layer_print_min[0]},{first_layer_print_min[1]} PRINT_MAX={first_layer_print_max[0]},{first_layer_print_max[1]}

```

**Klipper start code**

```
[gcode_macro PRINT_START]
description: Start code for all prints
gcode:
    POWER_ON_ALL
        # Use absolute coordinates
    G90
    # Start bed heating
    {% set BED_TEMP = params.BED_TEMP|default(75)|float %}
    #M140 S{BED_TEMP}
    M190 S{BED_TEMP}
    M109 S75
    M109 S0    
    # extruder relative mode
    #M83
    # Reset the G-Code Z offset (adjust Z offset if needed)
    #SET_GCODE_OFFSET Z=0.0
    # Home the printer
    #{% if printer.homed_axes != 'XYZ' %}
		G28			#Home All Axes
	#{% endif %}
    BED_MESH_CALIBRATE
    # Mesh profile
    #BED_MESH_PROFILE LOAD=default
    # Start Hotend heating
    {% set EXTRUDER_TEMP = params.EXTRUDER_TEMP|default(200)|float %}
    # Move the nozzle off the bed
    G1 Z10 F3000
    # Wait for bed to reach temperature
    M190 S{BED_TEMP}
    # Set and wait for nozzle to reach temperature
    M109 S{EXTRUDER_TEMP}
    # Move the nozzle very close to the bed
    #G1 Z0.15 F300
    G1 Z5 F240
    #Purge line
    G1 Z5 F500
    G1 X2.0 Y10 F3000
    G1 Z0.18 F500
    G92 E0 F500
    G1 X2.0 Y140 E10 F1500 ; prime the nozzle
    G1 X2.3 Y140 F5000
    G92 E0
    G1 X2.3 Y10 E10 F1500 ; prime the nozzle
    G92 E0            
    G1 Z2.0 F600        
    G1 Z0.2 F600        
    G1 Z50.0 F600
    G4 P5000 ;dwell for 5 seconds 
```

[OS1.6.1_PETG_21m41s.gcode.txt](https://github.com/SoftFever/OrcaSlicer/files/11175182/OS1.6.1_PETG_21m41s.gcode.txt)
[OS1.5.0_PETG_22m32s.gcode.txt](https://github.com/SoftFever/OrcaSlicer/files/11175183/OS1.5.0_PETG_22m32s.gcode.txt)

TexZeTech, 2023-04-07:
I'm going to try slicing this file in my new VM of Manjaro and try to replicate it there.

hliebscher, 2023-04-07:
i use {first_layer_temperature[0]} and [first_layer_bed_temperature]

SoftFever, 2023-04-07:
That is because your PRINT_START macro use different parameter names `EXTRUDER_TEMP` and `BED_TEMP`.
You need to change the machine start gcode to something like this:
`PRINT_START EXTRUDER_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=[bed_temperature_initial_layer_single]`

TexZeTech, 2023-04-08:
> That is because your PRINT_START macro use different parameter names `EXTRUDER_TEMP` and `BED_TEMP`.
> You need to change the machine start gcode to something like this:
> `PRINT_START EXTRUDER_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=[bed_temperature_initial_layer_single]`
> 

Welp this is definitely a PEBKAC moment. Thank-you for pointing that out. The fact that was working somehow as well 😑.

closed 2023-04-08 by TexZeTech