OrcaSlicer/OrcaSlicer #684: Imported objects are black and shading is impossible to see.

julie777, 2023-04-07:
**Describe the bug**

Black with black shading is a terrible choice for the default color of an object. 
I also cannot find a way to change the color of the object.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**

Import STL file

**Expected behavior**

Use a decent default color and provide a preference to change the object color.

**Screenshots**

The contrast actually makes it look better in the bug report because I am using github dark mode. OrcaSlicer opens in light mode so there is really no contrast on the object.

[image]

**Printer model**

**Desktop (please complete the following information):**
 - OS: Windows]
 - Version 1.6.1

hliebscher, 2023-04-07:
for me it takes the filament color. change from black to other ...

julie777, 2023-04-07:
Here is a screen shot. But now that you mention it, I remember that Bambu Slicer was showing the color next to the filament name in the chooser. I just went an looked and it was showing a color different from the color of the currently selected filament. Changing the filament selection made it correct itself.

I would still consider this an issue, because during the very first startup a filament is auto selected, but the color didn't get selected and used for the model.

[image]

After a little more digging, I determined that there are two bugs. 
1. During initial startup the default filament was not truly "selected" and its color used.
2. editing a filament definition (whether saved or not) does not update the filament color on the model object or next to the filament chooser like it should.

I will report the second bug to Bambu Slicer. The first did not happen when I set up bambu slicer, but did with Orca. Then when I changed my filament settings before printing the second bug kept the plate objects black. I am leaving the first bug for you to look into.

hliebscher, 2023-04-07:
pls click at the white square before the filament name 
select any other color 
then save the filament settings

for me that works fine

dcs3473, 2023-04-09:
That worked for me.  Like the default, color isn't attached

closed 2023-11-20 by github-actions[bot]