OrcaSlicer/OrcaSlicer #685: FEATURE REQUEST] Custom bed shape design

H4kunaM4tata, 2023-04-07:
For context:
Slice done with PrusaSlicer:

[image]

Slice done with OrcaSlicer:
I believe because there is no bed shape design set, the thumbnail looks so raw.

[image]

[image]

I did find the path to set it but it does not seem to accept the PNG file, nothing really happens.
The PrusaSlicer thumbnail looks a lot better, it is also better to see on the display and home assistant.
This is not the end of the world but would be nice to have this working.

[image]

Thank you

hliebscher, 2023-04-07:
you can add this self in your json config file... like ratrig

H4kunaM4tata, 2023-04-07:
> you can add this self in your json config file... like ratrig

I see, well, the settings is right there so it makes more sense to try get it working instead of workarounds. 
Workarounds get messy with time. After releases you might or might not have to manually change it again etc.

atl285, 2023-04-07:
@hliebscher, I've tried this with a Ender 3. But the changings in the json file were ignored.

hliebscher, 2023-04-07:
pls send the stl or svg as zip i integrate this for testing

atl285, 2023-04-08:
Here are the files for my Ender 3 (V1). I've got them from PrusaSlicer.
[ender3_files.zip](https://github.com/SoftFever/OrcaSlicer/files/11183314/ender3_files.zip)

SoftFever, 2023-04-18:
Fixed in 1.6.2 beta

closed 2023-04-18 by SoftFever

atl285, 2023-04-19:
Hi, if I load a custom Bed Shape in v. 1.6.2 it is possible, but the point (0,0) is set to middle of plate.
[image]
Is there a way to set manually the point (0,0) to the front left corner? Else any print will fail, because the front left corner is defined as negativ coordinate. I use the bed shape from my earlier comment.

sdylewski, 2025-09-26:
I have the latest orcaslicer and can't load a .png file of a bed image either.  This would be helpful to show a keep out region for stealtthchanger...