OrcaSlicer/OrcaSlicer #688: Inconsistent speed for overhanging

H4kunaM4tata, 2023-04-07:
I am trying to understand some differences between PS and OS.
OS overhanging for some reason does not have consistent slow speed.
It is a mix of slow with fast speed and that is messing things up,

[image]

Both slicers share the same setup

[image]
[image]

SoftFever, 2023-04-10:
Can you share the project file?

H4kunaM4tata, 2023-04-10:
@SoftFever definitely.

This is a piece of a bigger part, cut to size just to test the overhang stuff as I am setting OS with PS profiles.

[fan(8).stl.zip](https://github.com/SoftFever/OrcaSlicer/files/11188589/fan.8.stl.zip)

Thank you

jmlott, 2023-04-11:
I believe this is related to the bug I was about to open, so I'll just leave my comment here instead. In 1.6.1, it seems like something broke with overhang detection. If I slice a standard benchy with overhang detection enabled, it doesn't show the front bottom part of the boat as an overhang and therefore doesn't adjust for it. It was properly showing as an overhang in 1.5.0.

H4kunaM4tata, 2023-04-11:
> I believe this is related to the bug I was about to open, so I'll just leave my comment here instead. In 1.6.1, it seems like something broke with overhang detection. If I slice a standard benchy [...]

I see. I haven't used SoftFever for while after returning my X1C, I didn't even knew it had changed the name.
I have been moving away from PrusaSlicer to OrcaSlicer with my Voron ( bringing some folks along with me haha ) and have noticed it.
SoftFever is a lovely slicer, clean, organized and very active. Bugs get sorted out very quick which is awesome.

H4kunaM4tata, 2023-04-22:
Is there an ETA for a fix??

At the moment, I am being forced to go back to PrusaSlicer to get the prints to be printed properly
OS, left side printed just right, right side with greater overhang printed like...

[image]

What is this 170mm/s doing here??

[image]

Same model, same profile, different behavior.
Notice how the PS external walls keep a lower speed so better cooling, there is no fast speed mixed up.
OS external walls mixes fast and slow speed.

[image]

I have been using OrcaSlicer as my main slicer for my Voron but these bugs makes it harder to keep using it :(

H4kunaM4tata, 2023-04-22:
Both slicers are using Classic wall generator. Arachne does not fill gaps properly.

The only way OrcaSlicer will generate the walls with the right sleep is by setting it with Arachne which makes no sense.

H4kunaM4tata, 2023-04-22:
Edit: Giving OrcaSlicer another try.
I had `Detect thin walls ` disabled since 2 months ago when I still had my X1C.

This feature is or was broken, if a model had letters on it, it would just delete the whole letters and other crazy inconsistencies.
Giving this slicer another try, will report back.

R4toLoko, 2023-11-08:
Overhang logic has been improved and already released.

closed 2023-11-08 by R4toLoko