OrcaSlicer/OrcaSlicer #689: Mullticolor print without prime tower results in insanely long print time

lygris, 2023-04-07:
**Describe the bug**
Went to print a multicolor print today after upgrading to 1.6.1 and put in a filament change at a layer. Looking at the model print time after slicing shows35583992d72h49m. Trying to print it anyway the toolhead just stops at that layer change and sits there. tried to cancel the print but the printer remained unresponsive. using a prime tower fixing the time issue.

**3mf File for This Bug**
3mf zipped and uploaded
[bug.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/11175828/bug.3mf.zip)

**To Reproduce**
Steps to reproduce the behavior
on a layer in the prepare tab add a filament color change

**Expected behavior**
AMS pulls back current filament and changes to desired color. (the ams did change the filament, but the print didn't continue)

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Printer model**
Bambu X1C
**Desktop (please complete the following information):**
 - OS: macbook pro m2
 - Version latest macos

shlagwuk, 2023-04-07:
Same problem, voron printer, mac mini M1.
[image]

shlagwuk, 2023-04-25:
It seems like it's fixed on 1.6.2 beta
[image]

shlagwuk, 2023-10-14:
The bug is still here in 1.7.0
[image]

Print is stopped after the first fillament swap:
`Extrude only move too long (92233720368547.766mm vs 110.000mm)`
That's a lot of filament ;-)

closed 2024-02-19 by github-actions[bot]