OrcaSlicer/OrcaSlicer #690: Unable to parse the received print job

qgerome, 2023-04-07:
**Describe the bug**
I'm trying to print the Orca cube https://www.printables.com/model/444132-orca-cube/files using the latest version OrcaSlicer (1.6.1). When I send the print to my printer (Bambulab X1C @ 01.04.01.00) it failed after the upload to the cloud service with the error "Failed to start printing", "Unable to parse the received print job. Please update the firmware to the latest version and try again. [0500 4002 0954131]".

I tried the same 3MF file on BambuStudio and it works.

**3mf File for This Bug**
[OrcaCubev2_bug.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/11177161/OrcaCubev2_bug.3mf.zip)

**To Reproduce**
Load the 3mf file in OrcaSlicer (latest version or 1.5)
Send it to the printer
When the soft tries to change to the device page it shows the message that it failed

**Expected behavior**
It prints

**Screenshots**
[image]

**Printer model**
Bambulab X1C @ 01.04.01.00

**Desktop (please complete the following information):**
 - MacOS M2 Pro @ 13.2.1 
 - OrcaSlicer @ 1.6.1

SoftFever, 2023-04-07:
It is most likely a glitch in Bambu's cloud service.
Can you try again after a few minutes?

qgerome, 2023-04-07:
It's been like this since at least a week now :/ It's been already a few prints that I have to go back to Bambu Studio to print. Could I do something to give more logs ?

julie777, 2023-04-24:
It happens regularly to me and I use LAN only all the time. 
It must not be a cloud issue.

chuckkay, 2023-04-24:
> 

Did renaming the file fix the issue?

qgerome, 2023-04-24:
I updated to OrcaSlicer 1.6.2 and tested multiple things:

**The "faulty" file from this ticket**

It worked without problems

**A file with a URL encoded filename**

It failed with the same error as the one described here.

**The same file as above but with the filename corrected (no %20 in it)**

It worked.

Not sure if we can conclude that it is fixed or if we should fix it when it is sent to the cloud service.

chuckkay, 2023-04-24:
Definitely a bug Orca should be made aware of…

closed 2023-11-20 by github-actions[bot]